NorthernWidget/ALog #9: Streamline support for second interrupt

awickert, 2015-11-07:
Find a way to use the "wakeUpNow_tip" function to execute TippingBucketRainGage() more directly within the Logger class:

        // Check right before going back to sleep if there has been a rain
        // gauge bucket tip while it has been on
        // This is a temporary solution!
        if (NEW_RAIN_BUCKET_TIP){
          TippingBucketRainGage();
        }

emrtech, 2015-11-08:
Hi Andy,

I haven't played with this part yet. I expected to use any of the 
available interrupts to wake up the uC or use the hardware counter 
(which stays alive during sleep to increment the total. )

I had intended to use the T0/XCK (PB0) or T1 (PB1) inputs or INT2(PB2) 
for rain accumulation capture. Using INT2 should be safe too, if a 
software counter in RAM was used.

In my existing PIC based met station (which doesn't go to sleep) I was 
using an external INT input to catch rain gauge pulses. However, I have 
a debounce circuit ahead of the input. We need to be careful to debounce 
this input properly (HW or SW) to prevent switch bounce to falsely 
increment the count total. But so far far, I have put that part on the 
back burner. It might be useful to start thinking about the best 
implementation method. We have obviously choices. Using an INT without 
HW debounce would be OK if we were to disable the int with the main 
timer and re-enable it after 10+ms. This way multiple transitions would 
have no effect.

Next week I plan to test a number of brand new DELKIN industrial SD 
cards at work to make sure there is no hidden design problem in the 
logger circuit. As I have mentioned, all other cards work fine, other 
then the ("bad" marked company) cards.

I still have a mystery on my hand with U22. On all three logger boards, 
it gets warm when I run the BL1284P test program. when I run the sd card 
test program only, they stay cool. This one puzzles me;-) 
Troubleshooting has not revealed any stupid mistakes yet. I need to 
really knee myself into it.

Right now I am taking a short break with hobby stuff for a few days 
because of family related priorities and winter being on the door step, 
I have to get the house/garden ready for the new season.

Have a good weekend,
Gerhard

awickert, 2015-11-09:
Hi Gerhard (@emrtech),

The big challenge here is getting the functions to work both inside and outside the Logger class. The wake-up functions exist outside the class, and after these functions, the code continues to execute where it left off. Therefore, if a bucket tip occurs when the logger isn't asleep, I don't know where in the main code execution the MCU will be. There should be a way around this, and this ticket is mostly to remind myself to find such a way.

Software debouncing for rain gage, etc., interrupts with a 10 ms delay exists already in the code, so all set there.

If you want to discuss the U22 on GitHub, let's open a new project that contains your design. Ping me if you'd like me to do this for you. Good luck with continued tests and hope everything with family goes as smoothly as it can.

emrtech, 2015-11-10:
Hi Andy,

thanks for all your efforts on my behalf.

> Hi Gerhard (@emrtech <https://github.com/emrtech>),
>
> The big challenge here is getting the functions to work both inside
> and outside the Logger class. The wake-up functions exist outside the [...]
I think we need to test out those questions. As long as an external 
interrupt handler takes care of  the rain gauge count, it shouldn't 
matter what other code is running or if it is asleep.

The real question is, should we maintain the R.G count in RAM for a 
while and  update the SD card periodically or write it to EEPROM for 
safekeeping? The EEPROM is probably good for several million writes, 
especially if we use some kind of wear reduction algorithm. In my 
weather station I was using the internal RTC (DS1306 and now DS3234) 
SRAM to maintain the R.G. count for safe keeping. Because of battery 
backup it is quite safe in there, especially with the DS1306 which 
featured a write protect bit. In ten years, I never lost any data from 
teh DS1306.

The problem with the DS3231 is, it ain't got any SRAM;-) On the other 
hand we could use a couple of unused alarm registers to store rain gauge 
data. It would be safe from resets to the uC. Should the W.D bite, the 
RAM data inside the uC would be lost. It is a bit of an awkward problem.

There might also be a possibility to vector the watchdog reset to call 
an WD ISR and put in safekeeping code in there to preserve the data 
before the uC becomes reset. But, this may raise some concerns as well.

On one hand we need to minimize EEPROM/SDCard write accesses to preserve 
battery energy and yet to keep the data safe. On the other hand we could 
consider replacing the DS3231 with an DS3232 which contains extra 
protected RAM.

What approach would you recommend? Obviously, you had to confront this 
yourself as well.

Another possibility with reasonable success would be to write the rain 
data to SDCard whenever the logger is called upon to do so. You could 
lose at worst 10 minutes of rain count. That may be acceptable in low 
rain regions.

> Software debouncing for rain gage, etc., interrupts with a 10 ms delay 
> exists already in the code, so all set there.
I wouldn't expect otherwise;-)
>
> If you want to discuss the U22 on GitHub, let's open a new project 
> that contains your design. Ping me if you'd like me to do this for 
> you. Good luck with continued tests and hope everything with family 
> goes as smoothly as it can.
Yes, it would be nice if you could initially help out with it. Perhaps 
it is easier to add things then start things. I could send you some of 
the current design files. I have started a version 2.1 of the logger 
which adds another EEPROM and a pullup to the 32khz monitoring output of 
the RTC. I suppose it is easy to publish revised versions after the 
initial upload. As to my test software. I would like to wait a little as 
I am adding functionality to allow complete logger hardware testing. 
Should you want a snapshot anytime, just let me know. If you feel we 
should start putting the current version I am OK with that as long as 
adding revisions is not an issue. I just don't want too much unfinished 
construction code out in the world. Perhaps, I am seeing these things 
not properly.

I need to find out on how to add text myself. I am really embarrassed to 
be so afraid of communicating with Github. It really feels like poking a 
monster - you never know how it will react and strike out at you. How 
good is Github at undoing things and forget? If I can figure out on how 
to undo unwanted operations, it might put me to ease. Somehow, I get the 
impression Github never forgets any action and that might make it 
difficult to correct things.

A tested five DELKIN cards today at work. They all worked 100%. So, my 
bad card is "bad". It seems just a little voltage sensitive. If I power 
the card with a separate power supply at 3.35V it works perfectly too. I 
filled the card up with files on the PC and had no issues there.

Gerhard

awickert, 2015-11-10:
> I think we need to test out those questions. As long as an external 
> interrupt handler takes care of  the rain gauge count, it shouldn't 
> matter what other code is running or if it is asleep.

In principle, yes; in practice, if my sensor functions are inside the class, this will take some work. I am reading up on it.

>The real question is, should we maintain the R.G count in RAM for a 
>while and  update the SD card periodically or write it to EEPROM for 
>safekeeping?
> ...

This could be a workaround if we could write the time to the EEPROM outside of class functions (that include those for the rain gauge). Which actually keeps it as the original problem, I think. So I think that I would prefer to try to solve this problem within the object-oriented framework.

You bring up another set of points on internal vs. external (SD card) writes. This is good fodder for another ticket here, and I think would be well-worth some thinking, as it would require all of the functions to be re-tooled but could save a lot of power.

> [Current design and code files]

I think that having v. 2.1 up would be fantastic. Could you please send me to them so I can create the repository? The whole point of GitHub is that it is very easy to commit updates, so I think that it is best to put the in-progress code up and use GitHub to record progress. Many open-source projects are lost because their creators did not want to post unfinished code; my philosophy is that code is always unfinished in one way or another, and that it's better to share early and often, and with comments.

Github / git are really good at undoing things, and you can always force it to forget changes you might make. Don't worry, you're not going to start global thermonuclear war here :)

> A tested five DELKIN cards today at work. They all worked 100%. So, my 
> bad card is "bad". It seems just a little voltage sensitive. If I power 
> the card with a separate power supply at 3.35V it works perfectly too. I 
> filled the card up with files on the PC and had no issues there.

Good to know, but in case you didn't realize, your replies are part of a ticket on GitHub, so I'm going to really try to keep this on-topic.

awickert, 2015-11-10:
Found that the best solution to this involves creating a function within the actual Arduino sketch itself that calls the class function. However, not sure how it will work with the sleep/wake functionality. Makes sense to have it in the sketch that instantiates the class, though:
http://arduino.stackexchange.com/questions/4862/attachinterrupt-in-a-library

awickert, 2016-07-15:
Good discussion with Gerhard to archive. The rain gauge is working on the external interrupt pin now, and I think that the whole system is about as good as we can expect, at least for the moment. Not everything can be a class function when interrupts are concerned.

Essentially follows the stack exchange solution, except that it is nestled within the general wake up routine.

closed 2016-07-15 by awickert