NorthernWidget/ALog #14: Alarms at major intervals

awickert, 2016-11-29:
@superchap123 made a much more general set of alarm inputs, so we can now specify days (of week), hours, minutes, and seconds. However, now the alarms ring at a time that is (last alarm time + dt) instead of going off every minute, every 10 seconds, etc.

We want measurements at major times that are synchronous across a catchment. Therefore:

| Interval | Example                     |
|------------|-------------------------------------|
| 2 seconds  | 0, 2, 4, 6, ...                     |
| 10 seconds | 0, 10, 20, ...                      |
| 5 minutes  | 0, 5, 10, ...                       |
| 7 seconds  | 0, 7, 14, 21, 28, 35, 42, 49, 54, 0 |
| 5 hours    | 0, 5, 10, 15, 20, 24/0, 5, ...      |

Note that the last two do not divide evenly into minutes or days, respectively. These are poorly-chosen sampling intervals, my best idea for dealing with them will be to have them start at 0 every time the next-higher unit increments.

awickert, 2017-04-26:
@awickert mostly fixed this with commit 085b9b6b4da88eaeda68f6b9f05bffdafbfe7f0c.
@superchap123 then finished fixing, I think, with commit 4e1d8da99770becc41692e0698195a94394d23e7

@superchap123 are we good to close this one?

superchap123, 2017-04-26:
Yes

closed 2017-04-26 by superchap123